lcgamboa/picsimlab #94: crash when selecting open hex file

cnoam, 2023-08-27:
The program crashes consistently when selecting File | open Hex file

**To Reproduce**
Steps to reproduce the behavior:
1. Menu "File | Load Hex"
2.  crash

**Expected behavior**
Open dialog window to choose a file

**command I used**
$ ./PICSimLab-0.9.0_230729_Linux_x86_64.AppImage

**Desktop (please complete the following information):**
 - OS: ubuntu 22.04
 - Version 0.9

**log file: picsimlab_error0.txt**

(picsimlab:149743): Gtk-WARNING **: 08:58:12.240: Loading IM context type 'ibus' failed

(picsimlab:149743): Gtk-WARNING **: 08:58:12.242: /usr/lib/x86_64-linux-gnu/gtk-3.0/3.0.0/immodules/im-ibus.so: undefined symbol: g_memdup2

(picsimlab:149743): Gtk-WARNING **: 08:58:12.242: Loading IM context type 'ibus' failed

(picsimlab:149743): Gtk-WARNING **: 08:58:12.444: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
PICSimLab: Debug On=0  Type=GDB Port=1234
PICSimLab: Remote Control Port 5000
Spare parts: parts[00] (Step Motor) created 
Spare parts: parts[01] (Text Box) created 
Spare parts: parts[02] (IO Virtual Term) created 
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
PICSimLab Caught SIGABRT: usually caused by an abort() or assert()
PICSimLab Stack[00]: /lib/x86_64-linux-gnu/libc.so.6(raise+0x16) [0x7f74c0642476]
PICSimLab Stack[01]: /lib/x86_64-linux-gnu/libc.so.6(abort+0xd3) [0x7f74c06287f3]
PICSimLab Stack[02]: /tmp/.mount_PICSimqNIBSl/usr/lib/libglib-2.0.so.0(+0x1db43) [0x7f74c08e6b43]
PICSimLab Stack[03]: /tmp/.mount_PICSimqNIBSl/usr/lib/libglib-2.0.so.0(g_assertion_message_error+0xb6) [0x7f74c09444c6]
PICSimLab Stack[04]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x213bca) [0x7f74bfa13bca]
PICSimLab Stack[05]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2142d6) [0x7f74bfa142d6]
PICSimLab Stack[06]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x214394) [0x7f74bfa14394]
PICSimLab Stack[07]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2144e0) [0x7f74bfa144e0]
PICSimLab Stack[08]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x228191) [0x7f74bfa28191]
PICSimLab Stack[09]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x182417) [0x7f74bf982417]
PICSimLab Stack[10]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x18675e) [0x7f74bf98675e]
PICSimLab Stack[11]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x228987) [0x7f74bfa28987]
PICSimLab Stack[12]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9441) [0x7f74bfae9441]
PICSimLab Stack[13]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9ce6) [0x7f74bfae9ce6]
PICSimLab Stack[14]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(gtk_widget_get_preferred_width+0x86) [0x7f74bfae9dd6]
PICSimLab Stack[15]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x12fc6f) [0x7f74bf92fc6f]
PICSimLab Stack[16]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x182417) [0x7f74bf982417]
PICSimLab Stack[17]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x18675e) [0x7f74bf98675e]
PICSimLab Stack[18]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x130737) [0x7f74bf930737]
PICSimLab Stack[19]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9441) [0x7f74bfae9441]
PICSimLab Stack[20]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9ce6) [0x7f74bfae9ce6]
PICSimLab Stack[21]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(gtk_widget_get_preferred_width+0x86) [0x7f74bfae9dd6]
PICSimLab Stack[22]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9441) [0x7f74bfae9441]
PICSimLab Stack[23]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9ce6) [0x7f74bfae9ce6]
PICSimLab Stack[24]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(gtk_widget_get_preferred_width+0x86) [0x7f74bfae9dd6]
PICSimLab Stack[25]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x12a9b9) [0x7f74bf92a9b9]
PICSimLab Stack[26]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2c52ba) [0x7f74bfac52ba]
PICSimLab Stack[27]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9441) [0x7f74bfae9441]
PICSimLab Stack[28]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9ce6) [0x7f74bfae9ce6]
PICSimLab Stack[29]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(gtk_widget_get_preferred_width+0x86) [0x7f74bfae9dd6]
PICSimLab Stack[30]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x182417) [0x7f74bf982417]
PICSimLab Stack[31]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x18675e) [0x7f74bf98675e]
PICSimLab Stack[32]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2426eb) [0x7f74bfa426eb]
PICSimLab Stack[33]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9441) [0x7f74bfae9441]
PICSimLab Stack[34]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2e9ce6) [0x7f74bfae9ce6]
PICSimLab Stack[35]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(gtk_widget_get_preferred_width+0x86) [0x7f74bfae9dd6]
PICSimLab Stack[36]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x242402) [0x7f74bfa42402]
PICSimLab Stack[37]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x182417) [0x7f74bf982417]
PICSimLab Stack[38]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x18675e) [0x7f74bf98675e]
PICSimLab Stack[39]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x2425a8) [0x7f74bfa425a8]
PICSimLab Stack[40]: /tmp/.mount_PICSimqNIBSl/usr/lib/libgtk-3.so.0(+0x182417) [0x7f74bf982417]
PICSimLab Stack[41]: /tmp/.mount

lcgamboa, 2023-08-27:
This is a known problem with appimage. The gtk file dialog used by wxwidgets doesn't work with appimage in new Linux versions.  Can you try the "modern" appimage version [PICSimLab-0.9.0_230826_Linux_Debian_Test_x86_64.AppImage](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab-0.9.0_230826_Linux_Debian_Test_x86_64.AppImage) ?

cnoam, 2023-08-30:
> I have updated the appimage scripts. Now it work in my computer without crashing when the load/save menu is open.
> Can you try this updated version: PICSimLab-0.9.0_230827_Linux_x86_64.AppImage<https://protect.checkpoint.com/v2/___https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab-0.9.0_230827_Linux_x86_64.AppImage___.YzJlOnRlY2huaW9uOmM6bzozOGEwMzRjM2Q4NjBiNDJlNDhjNjkyYWUxZTc5MDcxZjo2OjVhZmU6M2VhOTdlYzFjZjc5ZTkyZmZkZDBhNmQyMTE1ZDFlODVlYTQwNjUxM2NiZTFmMDkzMmE4NjU2ZGQ4Y2MzM2FkMjpoOlQ> ?
>
>
> External e-mail, be judicious when opening attachments or links

version 230827 works fine!

closed 2023-09-02 by lcgamboa