lcgamboa/picsimlab #95: Can't run it in Manjaro (Arch) Linux

esbon125, 2023-08-27:
i installed it perfectly from AUR, and using ```makepkg -si```. The error I'm getting is:
``` 
$picsimlab
picsimlab: error while loading shared libraries: libwx_gtk3u_html-3.0.so.0: cannot open shared object file: No such file or directory
```
I tried reinstalling, restarting. Nothing happened. I saw from another application it might my wxconfig which is not working, but i checked and i have the html library, so I don't know.

lcgamboa, 2023-08-27:
I'm not familiar with `makepkg` command (I'm a Debian user).  The PICSimLab compiles and run with the build script 
`./bscripts/build_all_static.sh` ?

esbon125, 2023-08-28:
Hello, it doesn't use it at all, because the script uses apt, and in Arch you have pacman, the PKGBUILD script is as follows:
```# Maintainer: yjun <jerrysteve1101 at gmail dot com>
# Contributor: Dimitris Kiziridis <ragouel at outlook dot com>

pkgname=picsimlab-bin
_pkgname=PICSimLab
_pkgname2=${pkgname%-bin}
_date=230728
pkgver=0.9.0
pkgrel=2
pkgdesc="A realtime emulator of development boards with integrated MPLABX/avr-gdb debugger"
arch=('x86_64')
url='https://github.com/lcgamboa/picsimlab'
license=('GPL2')
provides=("${pkgname%-bin}")
conflicts=("${pkgname%-bin}")
depends=('gtkwave'
	 'gpsim'
	 'libgcrypt'
	 'cutecom'
	 'gedit'
	 'pulseview'
	 'glibc'
	 'dtc'
	 'gcc-libs'
	 'glib2'
         'picsim'
         'openal'
	 'pixman'
	 'libslirp'
	 'wxwidgets-common'
	 'wxgtk3'
	 'zlib')
makedepends=('tar')
# source=("${_pkgname}-${pkgver}.deb::${url}/releases/download/v${pkgver}/${_pkgname}_${pkgver}_${_date}_Ubuntu_20.04.5_LTS_amd64.deb")
source=("${_pkgname}-${pkgver}.deb::https://sourceforge.net/projects/picsim/files/latestbuild/${_pkgname}_${pkgver}_${_date}_Ubuntu_20.04.6_LTS_amd64.deb/download")
sha256sums=('425af9de1bb7e208d6a719b42d543ffa139d596b04c1db1d62a760ce110f33e4')

package() {
  # extract
  tar xvf data.tar.xz -C "${pkgdir}"

  # remove doc
  rm -rf ${pkgdir}/usr/share/doc/

  # icon
  for icon in espmsim picsimlab srtank
  do
    install -Dm644 ${pkgdir}/usr/share/${_pkgname2}/$icon.png -t ${pkgdir}/usr/share/pixmaps/
  done
}
```
I've tried to communicate with the package maintainer, but I haven't been lucky yet.

lcgamboa, 2023-08-29:
Apparently, the [package](https://aur.archlinux.org/packages/picsimlab-bin) is created from .deb of ubuntu 20.04. This .deb package doesn´t work in recent versions of Linux ( they use gtk3.2 instead of gtk3.0).
You can try the [Appimage](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab-0.9.0_230827_Linux_x86_64.AppImage)  when the AUR package is not updated.

esbon125, 2023-08-29:
Thanks! Appimage worked just fine. Will close the issue, but maybe leaving a warning for newcomers like me wouldn't hurt

closed 2023-08-29 by esbon125