lcgamboa/picsimlab #98: Is it possible to separate the backend simulator/emulator when compile the full package?

laris, 2023-11-13:
For example I only focus on the simavr and use peripheral spare parts and gtkwave etc. don't plan to use other MCU, is it possible to build a standalone package/app to only support AVR MCU?

Or even separate the backend and let them with standalone app to support picsimlab, like homebrew install simavr as backend and picsimlab find the simavr via config file?

lcgamboa, 2023-11-13:
Hi @laris ,

yes, PICSimLab is modular. You can edit src/Makefile .Common and comment on the lines that include all boards and add only the boards that you want. Like the example below.
 ```
#OBJS+= $(patsubst boards/%.cc,boards/%.o,$(wildcard boards/bsim*.cc)) 
#OBJS+= $(patsubst boards/%.cc,boards/%.o,$(wildcard boards/board*.cc))
OBJS+= boards/bsim_simavr.o boards/board_Arduino_Uno.o
```
You will still need to install the base libraries and simavr and picsim (only necessary for some definitions). You can edit and remove the ucsim and gpsim libraries from the main Makefile too. The qemu libraries do not need to be removed as they are only loaded by qemu-based boards. If you are using the build script you can remove all references to ucsim, gpsim and qemu to build your simplified version based only in avr.

laris, 2023-11-13:
Thanks, that maybe can let me make a minimal app in macOS to avoid some parts building error.
I will try it.

laris, 2023-11-15:
https://github.com/laris/picsimlab

forked version, only provide simavr backend and AVR boards
```
OBJS+= boards/bsim_simavr.o # simavr backend
OBJS+= boards/board_Arduino_Mega.o
OBJS+= boards/board_Arduino_Nano.o
OBJS+= boards/board_Arduino_Uno.o
OBJS+= boards/board_Franzininho_DIY.o
```

macOS build discussion
ref: https://github.com/lcgamboa/picsimlab/issues/69

closed 2023-12-07 by lcgamboa