lcgamboa/picsimlab #99: Can't set LCD1602 R/W pin to GND in gpboard.

eagl1, 2024-01-02:
Hi,

I tried to set the pin to GND but couldn't. I even sat it to RE0 and pulled the pin to 0V but also didn't work.

This is the picture of my simulation.

[image]

This is the code I used:

1. LCD source code:

```
#include <xc.h>
#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>
#include "lcd1602.h"
#include "config.h"

char val_arr[16];

void lcd_init(void){	
	__delay_ms(5);
    lcd_cmd(0x33);
    lcd_cmd(0x33);
    lcd_cmd(0x33);
    lcd_cmd(0x32);
    lcd_cmd(0x38);
    lcd_cmd(0x0f);
    lcd_cmd(0x01);
    lcd_cmd(0x80);	
}

void lcd_cmd(uint8_t cmd){
    PORTD = cmd;
    PORTE &= ~(1 << LCD_RS);       
    PORTE |= (1 << LCD_EN);
    __delay_us(10);
    PORTE &= ~(1 << LCD_EN);
    if(cmd == 0x01 || cmd == 0x02){
        __delay_ms(2);
    }
    else{
        __delay_us(200);
    }
}

void lcd_data(uint8_t dat){
    PORTD =   dat; 
    PORTE |=  (1 << LCD_RS);
    PORTE |=  (1 << LCD_EN);
    __delay_us(10); 
    PORTE &= ~(1 << LCD_EN);
    __delay_us(200);
}

void move_cursor (uint8_t row, uint8_t col){
    if(row == 0){
        lcd_cmd(FIRST_ROW + col);
    }    
    else if(row == 1){
        lcd_cmd(SECOND_ROW + col);
    }      
}  
    
void lcd_string_arr(char str[]){    
    char i;
    for(i=0; str[i] != 0; i++){
        lcd_data(str[i]);
    }
}

void lcd_string(uint8_t row, uint8_t col, char str[]){
    move_cursor(row, col);
	lcd_string_arr(str);
}

void lcd_integer(uint8_t row, uint8_t col, double val){
    itoa(val_arr, (int)val, 10);
    lcd_string(row, col, val_arr);    
}
```

2. Main code:

```
#include <xc.h>
#include "lcd1602.h"
#include "config.h"

void main(void) {

    pic_init();
    PORTEbits.RE0 = 0;
    lcd_init();
      
    lcd_string(0, 1, "Hello");   
}
```

The code is working in PICGenios and Mclab2.

lcgamboa, 2024-01-02:
Can you confirm that RE0 is configured as output? 
`TRISEbits.RE0 = 0 ;`

eagl1, 2024-01-02:
Yep, it's sat as output in `pic_init();` I'm configuring all ports in this function.

lcgamboa, 2024-01-02:
Can you send me the workspace (the .pzw file) of the project?

lcgamboa, 2024-01-03:
I have done some tests and the LCD works with gpboard without problems.

PICSimLab workspace and Project: [lcdgpsim.zip](https://github.com/lcgamboa/picsimlab/files/13815100/lcdgpsim.zip) I have used in my tests.

eagl1, 2024-01-03:
It's working, thank you so much man :)

I downloaded the latest release.

[image]

closed 2024-01-03 by eagl1

eagl1, 2024-01-03:
The problem here:

`ADCON1 = 0x0C;`

I totally forgot about AN5, AN6 and AN7 are on PORTA. But I didn't have any problem in my PICGenios code.