lcgamboa/picsimlab #101: gpboard isn't fast as other boards; like, Mclab2 and PICGenios.

eagl1, 2024-01-11:
Hi,

I'm running the same hex file in gpboard and the other boards; like, Mclab2 and PICGenios.

It runs ok with Mclab2 and PICGenios. But slow in gpboard.

This is the code:

[multitask_877A_sim.X.zip](https://github.com/lcgamboa/picsimlab/files/13907019/multitask_877A_sim.X.zip)

lcgamboa, 2024-02-15:
Actually, the use of gpsim is not as optimized as that of PICSim in PICSimLab. Unfortunately, I don't have much time to improve gpsim support currently.

eagl1, 2024-03-08:
I switched to X board, it's fantastic.

Thanks for the wonderful platform.

lcgamboa, 2024-04-07:
In the[ last unstable version](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild), I have improved the gpboard simulation speed.

eagl1, 2024-04-07:
OK, I will test it.

I really like the improvements in the 9.1 version.

I checked the link, you just released the 9.2, thank you so much for this wonderful application. I'm running your application in all the lab computers.