lcgamboa/picsimlab #102: PICGenios board with PIC18F4580, crash upon invoking TIMER2.

abhishekaudupa, 2024-02-01:
**PICSimLab crashes upon trying to use the TIMER2 module in PICGenios board with PIC18F4580 MCU.**

When I initialize the timer2 module and switch the timer on using TMR2ON bit, the program crashes. The line for which PICSimLab crashes is shown below.

```
#include <xc.h>

    void __interrupt() isr() {
        static unsigned int ctr = 0;
        if (TMR2IF) {
            if (ctr++ == 2500) {
                ctr = 0;
                PORTB = PORTB ^ 1;
            }
            TMR2IF = 0;
        }
    }

    void main(void) {
        TRISB = 0;
        PORTB = 0;

        GIE = 1;
        PEIE = 1;

        TMR2IF = 0;
        TMR2IE = 1;
        PR2 = 125;
        TMR2ON = 1;  //crash (doesn't crash for TMR2ON = 0)

        while (1);
        return;
    }
```
 - OS: Windows 10, Home 64-bit, 10.0, Build 19045.
 

[image]
[image]

lcgamboa, 2024-02-01:
Hi @abhishekaudupa ,

thank you for reporting the issue. The problem in Picsim was fixed in commit [f3cfb55](https://github.com/lcgamboa/picsim/commit/f3cfb55c062cc1a23a6d5e4ee59b85ad1214da47).

abhishekaudupa, 2024-02-02:
@lcgamboa 
Thank you and you're welcome.

closed 2024-02-04 by abhishekaudupa

lcgamboa, 2024-02-04:
Test version with TIMER2 error fixed :[PICSimLab_0.9.1_240204_win64_setup.exe]( https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab_0.9.1_240204_win64_setup.exe)