lcgamboa/picsimlab #104: PICGenios with PIC18F4580 analog pins improperly configured.

abhishekaudupa, 2024-02-10:
I'm running PICGenios board with PIC18F4580 MCU. I tried to use RA0 pin as analog input but then found that RA0 is not an analog pin in the simulator. (RA0 is analog as per the datasheet of PIC18F4580). I also noticed that RB3 is shown to be an analog pin while it is not multiplexed as an analog pin.

Because of this I'm not able to use Potentiometer 1 on the board.

Summary:
RA0 should be an analog multiplexed pin.
RB3 should not be an analog multiplexed pin.

Evidence: Check the pin viewer.
[image]

lcgamboa, 2024-02-11:
Hi  @abhishekaudupa ,

thank you for reporting the issue. The problem was in Picsim again. It was fixed in commit [11e2d0e](https://github.com/lcgamboa/picsim/commit/11e2d0e55d725529ddab7062be254971a5e843f7).

abhishekaudupa, 2024-02-11:
@lcgamboa thank you!

closed 2024-02-27 by lcgamboa