lcgamboa/picsimlab #106: Gpboard simulation failed

Kabak85, 2024-03-15:
Tried with 12F675 on Gpboard on picsimlab 0.9.1 Win64 (Windows 7 64).

When using an HEX where OSCCAL is not set ( value 0x3FFF) and the internal oscillator, the simulation seems not running.

The OSCCAL value is set at programming by reading the original value from PIC and rewrite it.  The simulator MUST verify and replace it, if necessary, by a valid value ( 0x3400 to 0x34FF ) when loaded.

Not tested on others boards or other pic.

Kabak85, 2024-03-17:
gpsim 0.32 had the issue too.

I made a ticket about it at the same time : https://sourceforge.net/p/gpsim/bugs/292/

lcgamboa, 2024-03-28:
As soon as I have time I will update PICSimLab to use the newest version of gpsim.

lcgamboa, 2024-04-07:
The [last unstable version](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild) was updated to use gpsim 0.32.1.

Kabak85, 2024-04-07:
Hi @lcgamboa

I confirm it solve the OSCCAL issue.

Thanks.

closed 2024-04-07 by Kabak85