lcgamboa/picsimlab #107: Simulator for PIC18F87J60 (PIC18F97J60)

rtek1000, 2024-03-19:
Hello,

  I would like to suggest that it might be useful to simulate a PIC18F87J60 version.

  The PIC18F87J60 (and others perhaps) has a very low cell endurance (100-1k) and could be a major concern for firmware development on this IC.

[image]

Note: On the official [website](https://www.microchip.com/en-us/product/pic18f87j60) of the PIC18F87J60 the documents that appear are from the PIC18F97J60.

> Feature rich PIC18 Microcontroller with an integrated 10Mbps Ethernet communications peripheral

[image]

Reference:
Datasheet page 441
http://ww1.microchip.com/downloads/en/devicedoc/39762f.pdf

Note 2: Before seeing the new PICSimLab update with greater support for other uC, STM32, ESP32, W5500, and also for Linux (which interested me more), I already suggested that Wowki add the PIC18F87J60, but they have no knowledge of PIC, and are focused on making a profit before making samples available.

https://github.com/wokwi/wokwi-features/issues/755

https://github.com/wokwi/wokwi-features/issues/755

lcgamboa, 2024-03-21:
Hi @rtek1000 ,

Technically adding the PIC18F87J60 to PICSimLab is not very complicated and I can implement it. But implementing the ethernet controller would require more time, which I currently don't have available.

PICSimLab is my hobby, Wokwi is Uri's source of income. If Wokwi were open source it would be easier to collaborate directly, but I understand Uri's position.

lcgamboa, 2024-03-24:
I have added the PIC18F67J60 (low-pin version of PIC18F87J60 ) in the last build version, without ethernet support for now.

closed 2024-06-15 by lcgamboa