lcgamboa/picsimlab #108: [Question] Weird Servo Motor Movement

d-graz, 2024-03-30:
I will in advance admit that I'm not a pro user of this very nice piece of software (as i started using it 2 weeks ago). I have noticed the following using `Servo.h` library:
- `myservo.write(0)` works great and brings the motor to the correct angle
- `myservo.write(180)` works great and brings the motor to the correct angle
- `myservo.write(90)` does not works as expected as the motor reaches approximately 75 degrees and seems to wobble a bit while doing so.

At the start i tough that there was a problem in my code or my installation of the simulator but then i checked out also the online one to find that the same problem persisted.
For example [here](https://lcgamboa.github.io/js/picsimlab.html?../picsimlab_examples/pzw/board_Arduino_Nano/atmega328p/servo_Knob.pzw) in one of the example available on the wiki if you set the potentiometer to 2.5V (half of his 0V-5V range) the servo motor should reach 90 degrees but as you can see this does not happen.
I'm not marking this as a bug because maybe it's a simulator limitation or I'm not aware of some physics law the simulator is applying.
I would be happy if someone could give me an explanation for this behavior.
Thank you in advance.

lcgamboa, 2024-03-31:
Hi @d-graz ,

This problem is caused by some limitations of the PWM implementation in [simavr](https://github.com/buserror/simavr) (used by PICSimLab to simulate AVRs). 
The wobble in position is caused by calling the myservo.write function with a small delay, increasing the delay to values greater than 100ms reduces the problem. Apparently the simavr's PWM does not have the PWM register masking implemented, every time the myservo.write function is used the PWM pin goes to zero immediately after setting the duty cycle, even if this has not changed. Using a PWM without delay causes the pin to always be at zero (for example, placing just one analogWrite inside the loop without delay).
Another problem is that the value placed as duty cycle also has an offset error in the current simavr implementation, so placing a voltage of 2.5 Volts at the example input does not generate a 1.5ms pulse (which places the servo at 90 degrees).
This problem has been on the to-do list to be resolved for some time, but I haven't had any free time to work on PICSimLab lately.

d-graz, 2024-04-12:
Hi @lcgamboa,

thank you for the very clear explanation.
In case i wish to contribute to those fixes you mentioned early (if i manage to have some free time 😞)  should i directly go to the simavr repo or does your application uses a particular branch of the simulator?