lcgamboa/picsimlab #109: Gpboard crash on loading hex

Kabak85, 2024-03-30:
Hello,

Version : 0.9.2 240302 Windows64_WC DIST on windows 7

When loading an HEX file with a running simulation cause software freezing.
When you stop the simulation with the "power" switch, it's seem to work fine.

I suggest to shutdown the simulation automatically before loading the HEX file.

lcgamboa, 2024-03-31:
Hi @Kabak85 ,

Currently, when a new .hex file is loaded, the simulation [ends](https://github.com/lcgamboa/picsimlab/blob/829fff566cfdbf79cd52ee56ad9b679b7d70ffa7/src/lib/picsimlab.cc#L1149) and a new one starts. I was unable to reproduce this issue. What microcontroller did you use? Could you provide an example of a .hex file that causes the problem?

Kabak85, 2024-03-31:
Hi @lcgamboa

Got this issue with 12F675 and 16F84. Not tested on other MCU. I put the appdata/roaming folder and the hex to load  in attachments.

Launch picsimlab, load the hex files and it's freeze. 
In Windows "Task manager", picsimlab consume an entire core.

I tried the appimage under linux, the issue not appear. May be platform/machine related 

[appdataroaming.zip](https://github.com/lcgamboa/picsimlab/files/14817221/appdataroaming.zip)
[Hex file.zip](https://github.com/lcgamboa/picsimlab/files/14817222/Hex.file.zip)

lcgamboa, 2024-04-07:
I have updated PICSimLab to use the last gpsim version 0.32.1.

closed 2024-08-26 by lcgamboa