lcgamboa/picsimlab #110: Gpboard simulation speed issue

Kabak85, 2024-03-31:
Hello,

Version : 0.9.2 240302 Windows64_WC DIST on windows 7.

Simulation speed is not the same between gpboard and the oscilloscope/Vterm.

In this simulation, the 16F84 periodically send the letter "X" at the rate of 1600 BPS but scope/Vterm see it 6 times faster (9600 BPS).
The same HEX file on gpsim with a 1600bps USART terminal seem works correctly.

[Test serial.zip](https://github.com/lcgamboa/picsimlab/files/14813607/Test.serial.zip)

lcgamboa, 2024-03-31:
Hi @Kabak85,

Yes, I have already detected this problem but I still don't know how to solve it. The simulation step code is defined in this [function](https://github.com/lcgamboa/picsimlab/blob/829fff566cfdbf79cd52ee56ad9b679b7d70ffa7/src/boards/bridge_gpsim.cc#L108C1-L114C1), but it apparently is not working correctly. This problem affects the serial and other peripherals. Creating a blink using a timer and another using a delay, the one using the timer runs twice as slow as it should and the one using the delay six times slower, so the problem is not simply adjusting the frequency. If I manage to solve the problem I will let you know.

Kabak85, 2024-03-31:
Hi @lcgamboa 

I know that on PIC, the oscillator frequency (Fosc) can :
- drive directly some timer
- be divided by 4 (Fosc/4) to provide 1 CPU cycle instruction 
- some operand use 2 CPU cycle
 
I'm not very good with C and pointer and without comments ... but if i understand, your code using a state named "mExecute2ndHalf".  But if it's the second cycles of some instruction, you miss a lot of event.

I see ClockPhase but don't understand exactly what is it. If this value represent clock oscillator edge, you will have have a more reliable base.
I saw also cycles related counter in gpsim code used for simulation time.

lcgamboa, 2024-04-07:
Hi @Kabak85 ,

this issue is fixed in the [last unstable release](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild). 
[Clearing the halt breakpoint every cycle ](https://github.com/lcgamboa/picsimlab/blob/b6a11f2b7964c922579a5a5ecb7fd16e956a1732/src/boards/bridge_gpsim.cc#L105)makes the peripherals work at the correct speed.

Kabak85, 2024-04-07:
Hi @lcgamboa

I confirm you solve the issue.

Thanks.

closed 2024-04-07 by Kabak85