lcgamboa/picsimlab #112: MCP23S17 not working as inpurs

rocksail, 2024-06-05:
**Describe the bug**
When trying to use the MCP23S17 spare part, I set pins as inputs, but cannot make them work - when I read from them the value does not change.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files**
If applicable, add a minimal source code together with a PICSimLab workspace (.pzw file) to help reproduce your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Troubleshooting:**
The simulation in PICSimLab consists of 3 parts:

- The microcontroller program
- Microcontroller simulation (made by [picsim](https://github.com/lcgamboa/picsim) and [simavr](https://github.com/buserror/simavr))
- Simulation of boards and parts

When a problem occurs it is important to detect where it is occurring.

One of the most common problems is the error in the microcontroller program. Before creating an issue, test your code on a real circuit (even partially) to make sure the problem is not there.

Errors in the microcontroller simulation can be detected using code debugging. Any instruction execution or peripheral behavior outside the expected should be reported in the project of simulator used ([picsim](https://github.com/lcgamboa/picsim) or [simavr](https://github.com/buserror/simavr)).

If the problem is not in either of the previous two options, the problem is probably in PICSimLab. A good practice is to send a source code together with a PICSimLab workspace (.pzw file) to open the issue about the problem.

lcgamboa, 2024-06-05:
Hi @rocksail ,

[Input mode is not yet implemented](https://github.com/lcgamboa/picsimlab/blob/1524251009c50c4be6204087edb384161d74b115/src/parts/other_IO_MCP23S17.cc#L395). I will try to add support as soon as possible. Do you have any example code for reading MCP pins that I can use for testing during development?

rocksail, 2024-06-06:
Hi,

Thanks for answering - been trying to make this work for 3 weeks - tried a
thousand different libraries :)
Will give you a more comprehensive sketch.
In the meantime, I think there is another problem - Multiple MCP23s17s can
only be addressed with a unique CS and not
with same CS and different HW addresses ( A0, A1, A2 ).
WHen I tried to address the extenders with HW addresses, all extenders
respond in the same way as the add=0 chip.

thanks

Rgds
Ruben

lcgamboa, 2024-06-06:
The support to address in SPI of  MCP23s17 is incomplete but is simple to fix.  Thank you for the return. I will put this issue on the to-do list.

rocksail, 2024-06-06:
Glad to help!

rocksail, 2024-06-06:
Just a query,
I came across the picsimlab v8.(2) update that states that
the spare part for MCP23017 was added, but could not find it.
Was this released ?
Rgds
Ruben

rocksail, 2024-06-06:
> Hi Ruben,
>
> the MCP23017 is partially implemented as a device
> (same code of 23S), but there is no part yet to MCP23017, only for
> MCP23S017. It's simple to be finished but will require more time to be
> tested.
>
> Regards,
> Luis Claudio.

Ok, thanks for the update.

lcgamboa, 2024-06-08:
Hi @rocksail ,

I have added support to read, address, and interrupts in MCP23S17 in the [latest build](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild).  I have tested with the Arduino Adafruit MCP23017 library.

rocksail, 2024-06-11:
Thanks, I'll check it out!

rocksail, 2024-06-11:
Hi again,
Looks like all is working fine now ..... wonderful ! ! ! and big thanks ! !
!
rgds
Vik

rocksail, 2024-06-11:
Hi again,

Sorry to bother you again.
But I am trying to use the keypad spare part ( originally I was trying to
use it with MCP23s17 - but then I tried to use it with arduino uno
directly) but cannot make it work.

It seems that I can read the Lines but not the columns.

Is there a known issues please ?

Regds

rocksail, 2024-06-11:
my bad .... disregard

lcgamboa, 2024-06-15:
I have added support to MCP23017 on the [Latest code build.](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild)

rocksail, 2024-06-17:
Awesome, thanks!

closed 2024-07-07 by lcgamboa