lcgamboa/picsimlab #113: RTC ds1307

rocksail, 2024-06-14:
**Describe the bug**
Hi - I am trying to make this spare part work with various libraries : 
The time i set  fine but when I read the time the clock is reset to 2070/1/1  1:0:0 after the call.

I suspect that the command : RTCLIB_WIRE.write(0); to request data from ds1307 is resetting the time.

thanks again

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files**
If applicable, add a minimal source code together with a PICSimLab workspace (.pzw file) to help reproduce your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Troubleshooting:**
The simulation in PICSimLab consists of 3 parts:

- The microcontroller program
- Microcontroller simulation (made by [picsim](https://github.com/lcgamboa/picsim) and [simavr](https://github.com/buserror/simavr))
- Simulation of boards and parts

When a problem occurs it is important to detect where it is occurring.

One of the most common problems is the error in the microcontroller program. Before creating an issue, test your code on a real circuit (even partially) to make sure the problem is not there.

Errors in the microcontroller simulation can be detected using code debugging. Any instruction execution or peripheral behavior outside the expected should be reported in the project of simulator used ([picsim](https://github.com/lcgamboa/picsim) or [simavr](https://github.com/buserror/simavr)).

If the problem is not in either of the previous two options, the problem is probably in PICSimLab. A good practice is to send a source code together with a PICSimLab workspace (.pzw file) to open the issue about the problem.

lcgamboa, 2024-06-14:
Hi @rocksail ,

checking the ds1307 code I discovered 2 problems. The current code does not support the 12h mode and the day of the week does not range from 1 to 7 as it should, but from 0 to 6. I am fixing these problems. 
Please provide some example code where the problem you described occurs to see if it works after the correction.

lcgamboa, 2024-06-15:
I have fixed some issues with DS1307 in the [Latest code build. ](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild)

rocksail, 2024-06-17:
Ok, thanks.
I'll check it out this morning

rocksail, 2024-06-17:
Hi,
I have tried the new release but still I got the same problem :

sketch :

#include <Wire.h>
#include <RTClib.h>

using namespace RSys;

DS1307 RTC;

void setup() {
Serial.begin(115200);
Wire.begin();
RTC.begin();

RTC.adjust(DateTime(__DATE__, __TIME__));
delay(1000);
printDate();
delay(1000);
printDate();
delay(1000);
printDate();
delay(1000);
printDate();
delay(1000);
RTC.adjust(DateTime(__DATE__, __TIME__));
printDate();
delay(1000);
printDate();
delay(1000);
printDate();
delay(1000);
printDate();
delay(1000);
}

//===============================================================================
//  Main
//===============================================================================
void loop() {
printDate();
delay(1000);
}

void printDate() {
DateTime rtcNow = RTC.now();
Serial.print(rtcNow.year(), DEC);
Serial.print('/');
Serial.print(rtcNow.month(), DEC);
Serial.print('/');
Serial.print(rtcNow.day(), DEC);
Serial.print(' ');
Serial.print(rtcNow.hour(), DEC);
Serial.print(':');
Serial.print(rtcNow.minute(), DEC);
Serial.print(':');
Serial.print(rtcNow.second(), DEC);
Serial.println();
//RTC.adjust(DateTime(__DATE__, __TIME__));
delay(240);
}

Serial output :
2070/1/1 1:0:42
2070/1/1 1:0:43
2070/1/1 1:0:44
2070/1/1 1:0:45
2024/6/17 8:2:39
2070/1/1 1:0:1
2070/1/1 1:0:2
2070/1/1 1:0:3
2070/1/1 1:0:4
2070/1/1 1:0:5
2070/1/1 1:0:6
2070/1/1 1:0:7
2070/1/1 1:0:9
2070/1/1 1:0:10
2070/1/1 1:0:11
2070/1/1 1:0:12
2070/1/1 1:0:13
2070/1/1 1:0:14
2070/1/1 1:0:16
2070/1/1 1:0:17
2070/1/1 1:0:18
2070/1/1 1:0:20
2070/1/1 1:0:21

Hopes this helps-
Rgds

lcgamboa, 2024-06-17:
HI @rocksail ,
your example works without problem in PICSimLab 0.9.1 and 0.9.2  using Arduino IDE 1.8.19 with Uno board .  Try the workspace in the annex. 
[rtc_test.pzw.zip](https://github.com/user-attachments/files/15874171/rtc_test.pzw.zip)

rocksail, 2024-06-19:
Hi
Something is very strange.....
I tried the workspace attached - this is the output
.2070/1/1 1:0:0
2070/1/1 1:0:1
2070/1/1 1:0:2
2070/1/1 1:0:3
2024/6/17 13:26:7
2070/1/1 1:0:1
2070/1/1 1:0:2
2070/1/1 1:0:3
2070/1/1 1:0:4
.
.
.
Help / about PICSimLab :
Version 0.9.2.240615 Windows64_WX DIST

What do you suggest pls ?

Regards
Ruben

lcgamboa, 2024-06-19:
Hi @rocksail ,

The problem only occurs with the Windows version of PICSimLab . It appears to be something related to the OS's time synchronization functions of DS1307 part. Now that I managed to reproduce the problem I will try to fix it. 
Thanks for the feedback.

lcgamboa, 2024-06-19:
Hi @rocksail ,

I fixed the error, it was a problem formatting the year internally in the DS1307. You can test with this [patched version](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab_0.9.2_240622_win64_setup.exe).

rocksail, 2024-06-20:
Hi,

Thanks for the update - tried the patch - it works ... wellll but there is
a bug .
in the workspace file you sent me to try ... that now works fine
But if you set the date time to 2023-12-31 23:59:40 and let it run
strange things happen ! !
Can you kindly try ?

Regards
Ruben

rocksail, 2024-06-20:
Got some more info :
setting the year to 2023 or less will set the RTC date to year-4
so if i set the year to 2023 it will actually set the arduino RTC date to
2019
2022 becomes 2108, 2021 becomes 2017 and so on.
Setting the year to 2024 works great though.

lcgamboa, 2024-06-20:
Hi @rocksail ,
I appreciate your support in testing the PICSimLab.  Yes, there was a bug in the year calculation again, but it is now fixed. I have updated the [patched version](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab_0.9.2_240622_win64_setup.exe).

rocksail, 2024-06-21:
Hi again
Thanks for the update BUT there is now a stranger issue !

This is what I did :
void setup() {
    RTC.begin();
    DateTime dtNow = { 2023,2,28,23,59,50 };
    RTC.adjust(dtNow);
    printDate(dtNow);
}

void loop() {
    printDate(RTC.now());
    delay(1000);//simulate a delay as if other tasks are running
}

it works fine first time round - but if you run it again you get :
2023/2/28 23:59:50
2023/2/28 23:59:50
2023/2/27 23:59:51
2023/2/27 23:59:52
2023/2/27 23:59:53

lcgamboa, 2024-06-21:
I couldn't reproduce this problem, but I found another one. When you open the PICSimLab the date is recorded correctly in the RTC, but when the reset button is pressed the date is no longer recorded. Apparently, after the reset, the first byte of I2C is always corrupted. The problem is in the Arduino's i2c code, this problem will take more time to be fixed.

rocksail, 2024-06-24:
Ok, thanks for the update.