lcgamboa/picsimlab #114: PICSimLab Not responding

rayhan-shhadeh, 2024-06-21:
I have the latest version of PICSimLab **0.9.1v** at first it was fine but after couple of weeks everytime I try to load the Hex file it stops responding.
what I tried: 

1. Reinstalling it many times
2. Try installing older versions 7.11v , 8.11v
3. Delete Cache folder

- error produced text:

```
PICSimLab: Version "0.9.1 231210 Windows64_WX DIST"
PICSimLab: Command Line: C:\Program Files\PicsimLab\picsimlab.exe 
[ALSOFT] (EE) WaitForSingleObjectEx error: 0x102
[ALSOFT] (EE) WaitForSingleObjectEx error: 0x102
[ALSOFT] (EE) WaitForSingleObjectEx error: 0x102
```

lcgamboa, 2024-06-21:
Hi @rayhan-shhadeh , 

This error is related to the OpenAL library used for audio support for PICSimLab. A common problem that causes PICSimLab to freeze is using serial communication with a com0com port without correctly configuring it. If this is the case, connect a serial terminal to the other port of com0com and the PICSimLab  will work again.

rayhan-shhadeh, 2024-06-23:
Actually that's true, I'm using tera term for serial and another bootloader software ( originally for our uni ) that checks for available ports and make the connection to read last byte in memory then upload the code to a hardware basic circuit contains PIC18F4620, my question is how to configure com0com port correctly?

rayhan-shhadeh, 2024-06-23:
### **I Fix it**   ✅ 
  The problems root was from com0com drivers on my laptop so I updated all of them then launch Com0com and put COM1 (or any available port) in **`Tera Term -> New connection -> Serial -> port,`** and another port like COM2 on CuteCom in **`PICSimLab -> Tools -> Serial Terminal`** and that fixes the issue, Thanks for helping.

note: after installing com0c0m and launch it my laptop keeps showing that I need to install Microsoft .NET v2.0 or higher to show. however, I already installed the last version so I fixed this by:

**1. Control panel -> Turn windows features on or off.**

[image]

**2. Turn on .NET Framework (include 2.0 or higher).**
[image]

**3. Return to com0com and configure it.**
[image]

_____________________________________
**### Finally the project working.**
[image]

closed 2024-06-24 by rayhan-shhadeh