lcgamboa/picsimlab #117: No way to customize qemu efuse and bin file paths?

gw8484, 2024-07-06:
Hello,

I am using the unstable build (PICSimLab-0.9.2_240630_Linux_x86_64.AppImage) running board ESP32_C3_DevKitC_02.  I would like to custom efuse and bin files but the qemu config window does not allow editing.  Is there a way to change the efuse and bin file paths?

Thanks

gw8484, 2024-07-07:
Thanks for the quick response.  My use case is mainly to use PICSimLab/QEMU for trying different efuse settings (e.g. flash encryption, secure boot, etc.) without the risk of bricking real HW.  It seems if the qemu config could be an optional part of the saved workspace file then it would be possible to support custom efuse and bin files.

I'll try overwriting the fixed files.  Thanks.

lcgamboa, 2024-07-07:
Good point. I've never tested flash encryption in PICSimLab, but as the efuse file is not included in the workspace file it probably won't work correctly. 
The workspace file is simple a .zip file (renamed to .pzw) with the configuration files and .bin file.

lcgamboa, 2024-07-07:
I did some tests and verified that by copying the .efuse file into the .pzw file, the .efuse will be used instead of the .efuse in the home folder. This could be a good solution for your tests, it's easy to build a script to save the .bin and .efuse files in a .zip file and rename it to .pzw.

closed 2024-08-26 by lcgamboa

gw8484, 2025-02-23:
Thanks for the info.  I just got back to this again and would like to confirm that it works.