lcgamboa/picsimlab #118: Blue Pill Board Wrong schematics

godsi, 2024-08-16:
Hello,

I am currently using version 0.9.2 of PICSimLab (PICSimLab_0.9.2_240803_win64), and I recently encountered an issue while testing the BLUE PILL. It seems there might be a discrepancy between the schematics and the actual circuit.

I am currently programming using registers. According to the schematics, when PC13 is set to LOW, the LED should light up. However, after setting PC13 to LOW in the program, the LED did not light up. I tried setting PC13 to HIGH, and unexpectedly, the LED light up. Could it be that there's an internal circuit error? Below are my code and the results.

Thank you for your help.
[image]
[image]
[image]

lcgamboa, 2024-08-18:
You are correct, the implementation in the simulator had the polarity reversed. 
The correction was made in commit 1346954. 
[Version 0.9.2_240818](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild) already contains the changes. 
Thanks for reporting the problem.

godsi, 2024-08-19:
Hi @lcgamboa 

I downloaded the latest version, and its work fine

Thanks

closed 2024-08-19 by godsi