lcgamboa/picsimlab #120: AppImages Crash: esp32-simulator

kassane, 2024-08-19:
**Describe the bug**

I use appimage from ArchLinux (x86_64/zen3) for convenience and practicality.
I have used the previous version and also 0.9.2_24395 (older builds) without any problems - before you making new Ubuntu-based appimages.

- PICSimLab-0.9.2_240818_Ubuntu_20.04.6_LTS_x86_64.AppImage (missing libqemu)
- PICSimLab-0.9.2_240818_Ubuntu_22.04.4_LTS_x86_64.AppImage (missing libqemu)
- PICSimLab-0.9.2_240818_Ubuntu_24.04_LTS_x86_64.AppImage (crash app)

**To Reproduce**

See preview screenshots!

**Expected behavior**

Works similar the previous version or any `0.9.2_xxxxx_Linux_x86_64.AppImage` (old appImages).

**Screenshots**

<details>
<summary>clean-config</summary>

[image]
</details>

<details>
<summary>switch to esp32-device</summary>

[image]
[image]
</details>

lcgamboa, 2024-08-19:
Thanks for reporting the issue. I'll check it out. 
Previous versions were based on Ubuntu 20.04.6 only, the new version based on 20.04.6 was supposed to work like the old one.

lcgamboa, 2024-08-20:
I fixed the build scripts and now the Qemu libraries are no longer missing in versions 20.04.6_LTS and 22.04.6_LTS. 
As for the crash of version 24.04.6_LTS I couldn't find any problem, but the ideal is to always use the oldest appimage version that works on your Linux for compatibility reasons.

kassane, 2024-08-20:
Now, my latest test with versions 20.04 and 22.04 are solved. I also tried 24.04 again, though the issue persists.
Tried debugging and didn't succeed in discovering the real cause of this problem.

Tip: please add a notice in the releases or readme about the compatibility.

lcgamboa, 2024-08-24:
I configured the scripts to generate only one appimage version (the oldest and most compatible one).

kassane, 2024-08-24:
> I configured the scripts to generate only one appimage version (the oldest and most compatible one).

Well done. Thanks!

closed 2024-08-24 by kassane