lcgamboa/picsimlab #122: A bug in the PWM of BLUE PILL

godsi, 2024-08-20:
Hello,

I'm currently using PICSimLab_0.9.2_240818_win64. Lately, I’ve been focusing on learning about the BLUE PILL, so I might have a few more questions than usual. I hope that’s alright.

This time, I've encountered an issue with the PWM functionality on the BLUE PILL. No matter how I adjust the settings, the frequency displayed on the oscilloscope within PICSimLab always shows as 5000Hz. I've tried various number for ARR and PSC that control for PWM Frequency, but its not work, the frequency remains consistently at 5000Hz on the oscilloscope. Below is my code and the formula I'm using.

By the way, I'm certain that it's not an issue with the TIMER. I previously wrote a program involving the TIMER, and it behaved as expected.

Thanks
[image]
[image]

godsi, 2024-08-22:
Thank you for letting me know about the current status. I appreciate all the hard work that has gone into developing PICSimLab. I look forward to future updates. Thanks again for your support and efforts!

closed 2024-08-22 by godsi

lcgamboa, 2024-08-24:
I added support for configuring the STM32 PWM frequency in the [test version](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild).

godsi, 2024-08-25:
I am currently testing with the latest version, and the good news is that the frequency is now adjustable! However, the frequency appears to be halved. For example, when I set it to 50Hz, the oscilloscope shows 25Hz, and when I set it to 100Hz, the oscilloscope shows 50Hz. Below is my program and example.
[image]
[image]

lcgamboa, 2024-08-25:
You are right. I have fixed the qemu-stm32 code and updated the binaries. Please download the latest version again.

godsi, 2024-08-26:
I’ve tried the updated version, and it’s running smoothly now. Thank you so much for your quick response!

I’m truly grateful for your hard work in creating PICSimLab. This software has been incredibly helpful for my learning.

Currently, I’m using the standard library for stm32 to study, and I’m planning to upload my completed learning materials to GitHub in the future, including the PZW files. I hope you don’t mind. Thanks again!

lcgamboa, 2024-08-26:
When you publish your materials, let me know and I'll put some in the examples repository. I recently added a Bluepill programming [example in Rust ](https://lcgamboa.github.io/picsimlab_examples/board_Blue_Pill.html#board_Blue_Pill_stm32f103c8t6_Bmp180_Rust)made by a PICSimLab user.