lcgamboa/picsimlab #123: Feature Suggestion: Multiple Controller Support

RootUser3885, 2024-08-29:
I am currently working on a project that has multiple arduinos communicating via i2c. The ability to connect two or more arduinos via pins would be a really useful feature. Few if any other sim allows such a thing.

lcgamboa, 2024-09-02:
I did some tests some time ago to implement communication between multiple instances of PICSimLab. The results were not very good. But for I2C it should work reasonably well. As soon as I have time I will try to implement this support.

RootUser3885, 2024-09-05:
Excited to see it. If I was a better programmer, I would try to contribute but I do more embedded than PC programs. I've talked with other developers that say its difficult to get processes to communicate. So I can definitely see how it could be difficult.
Also, you think this would make SPI and possibly UART communication possible?

lcgamboa, 2024-09-06:
Making communication between processes is not the problem, the problem is doing it quickly and synchronized. In I2C it is possible to send and receive bytes between processes, in the case of SPI it is necessary to do it bit by bit, which causes greater delays in the simulation. Implementing the transfer of small amounts of synchronized data between processes is inefficient, but it is what has to be done in the case of simulators.
In the case of UART, it is now possible to communicate between several instances of PICSimLab using a null modem emulator (com0com or tty0tty) as described in the documentation.