lcgamboa/picsimlab #125: BluePill, problems loading BIN file generated through PlatformIO (VSCode)

kr4fty, 2024-10-08:
**Describe the bug**
After trying to load a BIN file compiled by VSCode, PICSimLab closes instantly

Is this a bug in the way the bin is generated or am I missing something?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Board'
2. Click on 'Blue Pill'
3. Go to 'File'
4. Click on 'Load Bin'
5. Select 'firmware.bin'
6. See error

**Expected behavior**
The LED will start to blink.

**Screenshots**
[image]

**Files**

* platforio.ini
``` platforio.ini
[platformio]
default_envs = generic_hid
[env:generic_hid] 
platform = ststm32
board = bluepill_f103c8_128k
board_build.mcu = stm32f103c8t6
framework = arduino
upload_protocol = hid
build_flags = 
            -D PIO_FRAMEWORK_ARDUINO_ENABLE_HID
            -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC -Os

```
* main.cpp
``` main.cpp
#include <Arduino.h>
#ifdef LED_BUILTIN
  #undef LED_BUILTIN
  #define LED_BUILTIN PC13
#endif
#define DELAY_TIME 1000

void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
  Serial.println("Hello World From STM32 Over USB CDC");
  digitalWrite(LED_BUILTIN, HIGH);
  delay(DELAY_TIME);              
  digitalWrite(LED_BUILTIN, LOW);
  delay(DELAY_TIME);
}
```

* firmware.bin
[firmware.bin](https://github.com/user-attachments/files/17293558/firmware.bin.txt)

* Config Qemu
``` Config Qemu
qemu-stm32
-M stm32-f103c8-picsimlab-new
-drive file=/tmp/picsimlab-CQw3nU/picsimlab_workspace//mdump_Blue_Pill_stm32f103c8t6.bin,if=pflash,format=raw
-rtc clock=vm
-icount shift=auto,align=off,sleep=off
```

* picsimlab_log0.txt
``` picsimlab_log0.txt
PICSimLab: Console output redirected to file: "/home/usuario/.picsimlab/picsimlab_log0.txt"
PICSimLab: Version "0.9.2 241005 Linux64_WX Appimage"
PICSimLab: Command Line: /tmp/.mount_PICSimyneQl4/usr/bin/picsimlab Blue_Pill stm32f103c8t6 /home/usuario/Documentos/firmware.bin 
PICSimLab: Load Config from file "/home/usuario/.picsimlab/picsimlab.ini"
PICSimLab: Window position x=223 y=90
PICSimLab: Window scale  1.02 
PICSimLab: Using board "Blue Pill"
PICSimLab: Instance number 0
PICSimLab: Opening "/home/usuario/Documentos/firmware.bin"
osc: 8000000
reset is called!

PICSimLab: qemu-stm32 -M stm32-f103c8-picsimlab-new -drive file=/home/usuario/Documentos/firmware.bin,if=pflash,format=raw -rtc clock=vm -icount shift=auto,align=off,sleep=off                                                                                            
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=00000000 R01=0000006f R02=d1036823 R03=00000000
R04=00000000 R05=00000000 R06=00000000 R07=00000000
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=20005018 R14=080009e3 R15=00000000
XPSR=40000003 -Z-- A handler
FPSCR: 00000000
PICSimLab Caught SIGABRT: usually caused by an abort() or assert()
PICSimLab Stack[00]: /usr/lib/libc.so.6(gsignal+0x20) [0x7e0b4e44c120]
PICSimLab Stack[01]: /usr/lib/libc.so.6(abort+0xdf) [0x7e0b4e4334c3]
PICSimLab Stack[02]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(target_words_bigendian+0) [0x7e0b48f1a9db]
PICSimLab Stack[03]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xac6334) [0x7e0b48ec6334]
PICSimLab Stack[04]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(armv7m_nvic_set_pending+0x30) [0x7e0b48ec63b6]
PICSimLab Stack[05]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(arm_v7m_cpu_do_interrupt+0x184) [0x7e0b48e63cbd]
PICSimLab Stack[06]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xb63186) [0x7e0b48f63186]
PICSimLab Stack[07]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xb6390e) [0x7e0b48f6390e]
PICSimLab Stack[08]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xb6398d) [0x7e0b48f6398d]
PICSimLab Stack[09]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(cpu_exec+0x9c) [0x7e0b48f63a2b]
PICSimLab Stack[10]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(tcg_cpus_exec+0x55) [0x7e0b48f85234]
PICSimLab Stack[11]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xb867fb) [0x7e0b48f867fb]
PICSimLab Stack[12]: /tmp/.mount_PICSimyneQl4/usr/lib/picsimlab//qemu/libqemu-stm32.so(+0xd43892) [0x7e0b49143892]
PICSimLab Stack[13]: /usr/lib/libc.so.6(+0x9439d) [0x7e0b4e4a339d]
```

**Desktop:**
 - OS: Linux Manjaro
 - Version: 24.1
 - IDE: VSCode
 - PlatformIO Core 6.1.16 - Home 3.4.4

kr4fty, 2024-10-09:
Hi @lcgamboa, what would be the appropriate method to upload my firmware to the emulated board, Serial perhaps?

lcgamboa, 2024-10-09:
H @kr4fty, using the documentation configuration you mentioned above. For stm32, the platformio upload button can be used to upload to PICSimLab, and the `picsimlab_tool`  will be used. In the case of debugging, gdb already loads the compiled program to PICSimLab before running. 
It is not possible to use serial emulation for uploading as it is not implemented in Qemu.

closed 2024-10-10 by kr4fty