lcgamboa/picsimlab #126: Can't use the app

MartinDiazUPSIN, 2024-10-10:
When I attempt to switch to another board using the tool in PICSimLab, the application freezes and becomes unresponsive. I've tried using different versions, but the issue persists.

Steps to reproduce the behavior:

Open PICSimLab and load a workspace.
Click on the tool to switch to another board.
The application immediately freezes and becomes unresponsive.
The interface becomes unclickable, and I have to force close the application.
Expected behavior
The application should allow switching between boards smoothly without freezing, and the new board should load properly for simulation.

[image]

OS:  Windows 10
Version: all versions

lcgamboa, 2024-10-10:
Hi @MartinDiazUPSIN . 
Could you send me the workspace that causes the problem and the log messages? 
Without opening the workspace can you switch the boards?

MartinDiazUPSIN, 2024-10-11:
Specifically this workspace: [image: image.png]
And how can I change the board without using the workspace?
Btw, thank you so much, I really appreciate your help.

MartinDiazUPSIN, 2024-10-11:
I forgot to upload this image: [image: image.png]
Thank you so much, and have a good night.

lcgamboa, 2024-10-11:
Hi @MartinDiazUPSIN ,

You don´t upload the workspace (the .pzw file) and the image.

Simple open the PICSimLab and switch boards using the menu `Board` don´t work to you?

[image]

MartinDiazUPSIN, 2024-10-17:
Hi,
That's exactly the problem when i try to do that, instantly this happens:
 bug.mp4
<https://drive.google.com/file/d/1AYKsMCe42AJl4Fb2Ppb9rsDV05iP3wxS/view?usp=drive_web>
Btw, thank you so much, it's nice that you are still answering and helping.

lcgamboa, 2024-10-18:
Hi @MartinDiazUPSIN , it's impossible to access the video you annexed without permission.

lcgamboa, 2024-10-18:
Try to delete PICSimLab memory files in dir "C:\Users\username\AppData\Roaming\picsimlab\"

MartinDiazUPSIN, 2024-10-25:
> Ok, I will try.
> Thanks again

Hi, that just worked, thank you so much.
Can you help me with something else?

lcgamboa, 2024-10-25:
Great. What do you need?

MartinDiazUPSIN, 2024-11-01:
First of all, I would like to express my sincere gratitude for the guidance
you offered me earlier. Your insights were incredibly helpful and I deeply
appreciate the time and effort you put into supporting me.

As a Mechatronics Engineering student with a desire to make significant
contributions in the field, I am currently in search of an internship
opportunity. My program requires me to secure a placement by November 15,
2024 and I am truly eager to work and grow within a professional
environment. I would be honored if there was a possibility to intern under
your guidance or with someone you know who can help me, from November to
January 2024.

Thank you very much for considering my application. I am attaching my CV
for your review and I will gladly provide you with any additional
information you may need.

Kind regards, from me (Martin Gonzalo Sanchez Diaz from "Universidad
Politécnica de Sinaloa").

closed 2024-11-17 by lcgamboa