lcgamboa/picsimlab #128: Error: EXCEPTION_ACCESS_VIOLATION

Bhuvanesh211, 2024-10-14:
PICSimLab: Console output redirected to file: "C:\Users\Admin\AppData\Roaming\picsimlab/picsimlab_log0.txt"
PICSimLab: Version "0.9.2 241005 Windows64_WX DIST"
PICSimLab: Command Line: C:\Users\Admin\Downloads\MICROCONTROLLER\PicsimLab\picsimlab.exe 
PICSimLab Error: EXCEPTION_ACCESS_VIOLATION

PICSimLab: Error closing PICSimLab in last time! 
Using default mode.
 Erro log file: C:\Users\Admin\AppData\Roaming\picsimlab/picsimlab_error0.txt
 If the problem persists, please consider opening an issue on github :
https://github.com/lcgamboa/picsimlab/issues

A backup was made of your project that presented a problem with the name following the template:
C:\Users\Admin\AppData\Local\Temp/backup_<boardname>_<timestamp>.pzw

lcgamboa, 2024-10-14:
Please add more information so I can reproduce and fix the error.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files**
If applicable, add a minimal source code together with a PICSimLab workspace (.pzw file) to help reproduce your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Troubleshooting:**
The simulation in PICSimLab consists of 3 parts:

- The microcontroller program
- Microcontroller simulation (made by [picsim](https://github.com/lcgamboa/picsim) and [simavr](https://github.com/buserror/simavr))
- Simulation of boards and parts

When a problem occurs it is important to detect where it is occurring.

One of the most common problems is the error in the microcontroller program. Before creating an issue, test your code on a real circuit (even partially) to make sure the problem is not there.

Errors in the microcontroller simulation can be detected using code debugging. Any instruction execution or peripheral behavior outside the expected should be reported in the project of simulator used ([picsim](https://github.com/lcgamboa/picsim) or [simavr](https://github.com/buserror/simavr)).

If the problem is not in either of the previous two options, the problem is probably in PICSimLab. A good practice is to send a source code together with a PICSimLab workspace (.pzw file) to open the issue about the problem.

lcgamboa, 2024-10-18:
Try to delete PICSimLab memory files in dir "C:\Users\username\AppData\Roaming\picsimlab"

closed 2024-11-17 by lcgamboa