lcgamboa/picsimlab #132: Picsimlab 0.92, not communicating with MPLAB X

Arto1234, 2024-11-27:
My system:
- Ubuntu 22.04.1 LTS 64 bit (6.8.0-49-generic)
- Picsimlab 0.92
- Board in Picsimlab is Mclab2 PIC16F877A
- tty0tty is loaded, checked with lsmod, but nobody is using it

Picsimlab config: serial port: /dev/tnt2
What may be wrong?

Arto1234, 2024-11-27:
tty0tty version is 1.4

lcgamboa, 2024-11-27:
Hi @Arto1234 ,
It is not necessary to configure tty0tty to communicate with MPLABX. tty0tty is for serial communication, if the objective is only to view the information sent by the serial port, it is simpler to use the spare part [Virtual Term](https://lcgamboa.github.io/picsimlab_docs/stable/IOVirtualTerm.html). To communicate with MPLABX it is necessary to install a [plugin](https://github.com/lcgamboa/picsimlab/releases/download/v0.9.2/com-picsim-picsimlab.nbm). The MPLABX debug plugin (com-picsim-picsimlab.nbm) don't work with MPLABX versions greater than 6.05

Arto1234, 2024-11-27:
Thanks for info, so quickly! :) Is fix coming in near future? I started to use MPLAB X 6.20...

lcgamboa, 2024-11-27:
As soon as Microchip updates your documentation.  See https://github.com/lcgamboa/picsimlab/discussions/124

Arto1234, 2024-11-28:
Thank you. So until then we have to use version 6.05.
I'll close this.
(I'd like to have your comment on another topic: 
What do you think, how big effort is to add simulation support for e.g. SAMV71 eval board?)

closed 2024-12-02 by Arto1234

lcgamboa, 2024-12-06:
The SAMV71 is a very complex device. Using the Qemu arm, it is possible to emulate the CPU, but the peripherals would all have to be implemented, which would take a lot of time.