lcgamboa/picsimlab #136: Windows Defender Flags PICSimLab 0.9.2 as a Trojan

BlitzdUK, 2025-01-31:
Hello,

I've downloaded the Windows setup file for PICSimLab v0.9.2 from the releases page:

🔗 [PICSimLab_0.9.2_241005_win64_setup.exe](https://github.com/lcgamboa/picsimlab/releases/download/v0.9.2/PICSimLab_0.9.2_241005_win64_setup.exe)

However, Windows Defender is detecting it as a threat and blocking it from running. It flags the file as containing:

🔍 Threat: [Trojan:Win32/Wacatac.H!ml](https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3AWin32%2FWacatac.H!ml&threatid=2147814523)

Could you please verify if this is a false positive or if the file needs to be rechecked? Let me know if you need any further details.

Thanks!

lcgamboa, 2025-01-31:
Hi,

this is a false positive as said in discussion https://github.com/lcgamboa/picsimlab/discussions/135 .

In the VirusTotal is false positive for 5 of 70 anti virus https://www.virustotal.com/gui/file/498ff65ddc4fe5374ffbf0f10008f943a1478927b1a5209c827a9ebc6776153b

lcgamboa, 2025-02-01:
MS answer:  
[image]

closed 2025-06-21 by lcgamboa