lcgamboa/picsimlab #137: Problem building 0.9.2 on WSL

kjellc, 2025-02-16:
Hello,

I'm trying to build the new 0.9.2 release of picsimlab, but it fails.
I have not been able to work around the problem.
I have been using the older 0.9.1 release for a long time without problems.

The error is strange since the compiler complains about not including <mutex>, but it is already included!
I'm running WSL2 on windows11.
The gcc version is 11.4.0

This is how I build:
  ```
git clone https://github.com/lcgamboa/picsimlab.git
  cd picsimlab
  bscripts/build_w64.sh
```

It stops with the following error:
```
In file included from ppicsimlab.cc:26:
picsimlab1.h:209:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  209 |     std::mutex cpu_mutex;
      |          ^~~~~
picsimlab1.h:57:1: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
   56 | #include "lib/picsimlab.h"
  +++ |+#include <mutex>
   57 |
picsimlab1.h:210:10: error: ‘condition_variable’ in namespace ‘std’ does not name a type
  210 |     std::condition_variable cpu_cond;
      |          ^~~~~~~~~~~~~~~~~~
picsimlab1.h:57:1: note: ‘std::condition_variable’ is defined in header ‘<condition_variable>’; did you forget to ‘#include <condition_variable>’?
   56 | #include "lib/picsimlab.h"
  +++ |+#include <condition_variable>
   57 |
make[1]: *** [Makefile.cross:37: ppicsimlab.o] Fel 1
make[1]: Lämnar katalogen ”/mnt/c/Devel/pic/picsimlab-0.9.2/src”
```
 
NOTE: I have tried compiling with -D_NOTHREAD, but then I get a different error.

lcgamboa, 2025-02-17:
Hi @kjellc ,

Apparently you are using an old version of gcc/g++ that does not fully implement native support for mutex and condition_variable. I use version 14 on Linux and version 13 on Mingw, which works without errors or warnings.

kjellc, 2025-02-17:
Thanks for a quick reply. I was using Ubuntu 22.04 LTS, and it's difficult to upgrade gcc.
So I started an upgrade to Ubuntu 24.04 LTS, which has gcc 13.3.0.
I have tried to build picsimlab with gcc 13.3.0 and it works fine. Just a couple of warnings.
Thanks for your help.

closed 2025-02-17 by kjellc