lcgamboa/picsimlab #140: VTerm Bug

PengJun1992, 2025-03-08:
**Describe the bug**
The STM32F103C8T6 outputs data to the UART serial port at a baud rate of 115200. However, the VTerm in picsimlab cannot display the data correctly when set to a baud rate of 115200, while it can display the data properly when the baud rate is set to 9600.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files**
If applicable, add a minimal source code together with a PICSimLab workspace (.pzw file) to help reproduce your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Troubleshooting:**
The simulation in PICSimLab consists of 3 parts:

- The microcontroller program
- Microcontroller simulation (made by [picsim](https://github.com/lcgamboa/picsim) and [simavr](https://github.com/buserror/simavr))
- Simulation of boards and parts

When a problem occurs it is important to detect where it is occurring.

One of the most common problems is the error in the microcontroller program. Before creating an issue, test your code on a real circuit (even partially) to make sure the problem is not there.

Errors in the microcontroller simulation can be detected using code debugging. Any instruction execution or peripheral behavior outside the expected should be reported in the project of simulator used ([picsim](https://github.com/lcgamboa/picsim) or [simavr](https://github.com/buserror/simavr)).

If the problem is not in either of the previous two options, the problem is probably in PICSimLab. A good practice is to send a source code together with a PICSimLab workspace (.pzw file) to open the issue about the problem.

lcgamboa, 2025-03-08:
Hi @PengJun1992 ,

This limitation has been removed in version 0.9.2. If you are using this version, please send me an example code so I can try to reproduce the error.

PengJun1992, 2025-03-11:
> Hi [@PengJun1992](https://github.com/PengJun1992) ,
> 
> This limitation has been removed in version 0.9.2. If you are using this version, please send me an example code so I can try to reproduce the error.

CODE：

          uint8_t ch;

	  if (HAL_UART_Receive(&huart1, &ch, 1, 10) == HAL_OK)
	  {
		  HAL_UART_Transmit(&huart1,&ch,1,HAL_MAX_DELAY);
	  }