lcgamboa/picsimlab #142: Blue Pill in PicSimLab crashes when enabling DMA

MichaelShu0920, 2025-03-13:
I used STM32CubeIDE to generate code for the STM32F103C8T6 and enabled DMA. After compiling it into a bin file and opening it in PicSimLab, the program crashes immediately. What could be the issue?

Thank you very much for your work, and I look forward to your reply.

This is the error report:
[picsimlab_error0.txt](https://github.com/user-attachments/files/19224044/picsimlab_error0.txt)

This is a simple setting:
[image]

closed 2025-03-14 by MichaelShu0920

lcgamboa, 2025-03-14:
Hi @MichaelShu0920 ,

Did you manage to solve the problem? What was the cause?

MichaelShu0920, 2025-03-14:
Thank you for your reply, so the issue lies with the qemu-stm32 used by PICSimLab, right? I think I need to research qemu-stm32 first, but the URL you provided  (http://beckus.github.io/qemu_stm32/) are inaccessible.

lcgamboa, 2025-03-14:
The qemu_stm32 fork is no longer being updated. I use another modified fork for PICSimLab (also not very updated). Below is the link to the line of code that detects the problem:

https://github.com/lcgamboa/qemu/blob/51df12c2106b844c232a6b184cf4384f5b1a7d37/hw/arm/stm32_dma.c#L371

I didn't write the DMA code. It may be wrong, or the code is reading an undocumented register.

MichaelShu0920, 2025-03-14:
Do you have time to fix the qemu_stm32 issue?
I closed the issue as it seemed unrelated to PICSIMLab.

MichaelShu0920, 2025-03-14:
Out of admiration for this project, if you have no time, I can start researching and make some contributions within my capabilities.