lcgamboa/picsimlab #146: New release not working after installation

eagl1, 2025-06-04:
Hi,

I get this error message after installing the software:

[image]

lcgamboa, 2025-06-04:
I tested it on three different computers with Windows 11 64-bit, and it worked on all of them. I tested both the 64-bit and 32-bit versions. I need more information to be able to reproduce the problem.

eagl1, 2025-06-04:
I downloaded this version:

[image]

I uinstalled the previous version and installed the new one, and when I launch the application, it gives me this error.

eagl1, 2025-06-04:
OK, I downloaded the .zip version and it worked with no issues.

So the .zip version works ok but the .exe isn't working and gives that error.

I installed the .exe version on my laptop too and it has the same error issue.

lcgamboa, 2025-06-05:
Hi @eagl1, I don't detect the problem in my computers, but I recompiled and updated the installer version. Can you try the update installer?

closed 2025-06-05 by eagl1